ricki-z/SDS011 #2: Noob needs help with sleep function

Druidus, 2017-03-26:
Could you please provide a short example how the sleep function is called. Thanks in advance!

ricki-z, 2017-03-26:
for sleep you can use the sleep() and wakeup() functions.
SDS011 mySDS(pinRX,pinTX);
mySDS.begin();
mySDS.sleep();
mySDS.wakeup();

Any command sent to the SDS011 will wake him up. So if you try to read data from a "sleeping" SDS011 that will wake up the SDS.

Druidus, 2017-03-27:
Thx for your answer! Helped me to track down the problem. 
`// Sleep
  my_SDS.sleep();
ESP.deepSleep(sleepTimeS * 100000);`

the deep.sleep() command for the ESP always wakes up the SDS011 - dont know who is to blame for that ;-) 
Thx again

cyrinux, 2017-04-11:
@Druidus I can confirme I have the same problem on a arduino mega with   Sleepy::loseSomeTime(sleepTime);

ubergesundheit, 2017-04-21:
@Druidus The issue with the deepSleep of the ESP8266 could be related to the fact, that the ESP pulls all pins to high just before sleeping. Since the SDS011 is connected through a SoftSerial, it could be possible, the SDS011 interprets this signal and wakes up.