ricki-z/SDS011 #3: About calculation of value

cyrinux, 2017-04-11:
Hi,
I read in the datasheet this:
PM2.5 value: PM2.5 (μg /m3) = ((PM2.5 High byte *256) + PM2.5
low byte)/10
PM10 value: PM10 (μg /m3) = ((PM10 high byte*256) + PM10 low
byte)/10

Could you please confirm me that you use this calcul in the library?

Thanks for your library :)

ricki-z, 2017-04-11:
This calculation is used in the library. Do you get any strange results?

cyrinux, 2017-04-11:
No I think there are good but just have a doubt sorry. I try to understand
where was this in the code... I'm bad cpp dev ^^ Ty

ricki-z, 2017-04-11:
I will try to explain ;-)
Look AT the  funktion read(), the switch statement.
Case 2 saves the high byte for pm2.5, case 3 shifts that value by 1 byte (equals multiply by 256) and adds the low byte. Cases 4 and 5 do the same for pm10. If all 10 bytes are receiver and the checksum is okay the saved values are converted to float and divided by 10.

cyrinux, 2017-04-11:
Thanks you very much for this explain. :)

closed 2017-05-01 by ricki-z