ricki-z/SDS011 #8: sleep and wakeup command

sisamiwe, 2018-03-10:
Hi,

how does the sleep and wakeup command shall be used?

thanks for reply

ricki-z, 2018-03-10:
Could you please look at the example?

sisamiwe, 2018-03-11:
Ok. 
my_sds.sleep() turn the sds into sleep mode and my_sds.wakeup() wakes it up.
What happens in sleep mode? Is the sensor totally of?

ricki-z, 2018-03-12:
The SDS011 will go down to < 4mA in sleep mode. For more see https://inovafitness.de/wp-content/uploads/2018/02/SDS011laserPM2.5sensorspecification-V1.3.pdf

sisamiwe, 2018-03-12:
OK, thanks.
How would you implement the sleep and wakeup command into your sketch example to have a measurement each 180s and then turn to sleep mode?

ricki-z, 2018-03-14:
Really? This should be a simple task if you think you can write own software.
You can look at the code of the ['airrohr-firmware'](https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/airrohr-firmware.ino). This code is reading 20 values from the SDS011 every 150 seconds.

rogalskj92, 2018-06-24:
Hi, I'm working on a Arduino Ethernet weather station which uses SDS011. Sensor is reading pm25 and pm10 values just fine, however I'm unable to put it into sleep mode. Even using premade schetches from SDS011 library in Arduino 1.8.5 I get the same results ( the fan is still spinnig and sensor doesn't go down to 4 mA). Do you know what I might be doing wrong? Thanks for reply.

ricki-z, 2018-07-01:
@rogalskj92 please check your connections between the Arduino and the SDS011. The sensor can be read even if no command can be send. This may be caused by a wrong connection between the RX pin on the SDS011 and the TX pin on the Arduino.