ricki-z/SDS011 #10: sleep() command not reliable

smuehlst, 2018-04-01:
The sleep() command appears to be unreliable. Every now and then calling the sleep() function does not stop the SDS011 sensor, and the fan keeps blowing.

I run  the following test program on an Adafruit Huzzah with an SDS011 sensor attached  to GPIO 4 and 14. The program turns on the sensor, waits 5 seconds, reads the results and puts the sensor to sleep for 5 seconds. When this program is run and watch the serial output, I see within a minute or two several occurrences that the fan keeps running althoug sleep() has been called:

	#include <SDS011.h>
	
	SDS011 my_sds;
	
	void setup() {
		my_sds.begin(4, 14);
		Serial.begin(115200);
	}
	
	void loop() {
		Serial.println("Wake up, measure for 5 seconds...");
		my_sds.wakeup();
		delay(5000);
		float p10, p25;
		while (!my_sds.read(&p25, &p10))
		{
			Serial.println("Repeat read...");
			delay(100);
		}
		Serial.println("P2.5: " + String(p25));
		Serial.println("P10:  " + String(p10));
		delay(2000);
		Serial.println("Go to sleep...");
		my_sds.sleep();
		Serial.println("Sleep 5 seconds...");
		delay(5000);
	}

5qm, 2018-04-17:
I had the same problem. After delete all "Serial.xxx" run the sleep mode:

```
`#include <SDS011.h>

float p10,p25;
int error;

SDS011 my_sds;

void setup() {
	my_sds.begin(0,1);
	//Serial.begin(9600);
}

void loop() {

my_sds.sleep();

	delay(1000);
}
````

smuehlst, 2018-04-29:
@5qm: Thanks for the suggestion. I tried this, and even if I comment out all operations on the Serial object in my test program it is still reproducible that once in about every 10 cycles or so the sensor fan keeps running until the next invocation of `my_sds.sleep();`.

ricki-z, 2018-07-01:
The sleep() function doesn't read the answer of the SDS011 (to minimize used memory). But in some cases the SDS011 is sending data while the sleep command is transmitted. It seems like the SDS011 can't receive a command while sending data.

MrAndroidGeek, 2019-04-20:
Did anyone get a solution working for this? I have the sds011 returning readings for 30 seconds every 15mins and often get a period where the sleep command fails to work.