ricki-z/SDS011 #14: Particle concentration calculus

mcanu44, 2018-06-21:
Hi,

Why do you not follow the particle concentration calculus proposed by the manufacturer in the datasheet ?
The manufacturer proposes to calculate the concentration from MSB and LSB as : MSB*256+LSB/10 (page 8).
You do MSB*256+LSB in your code: this gives an overestimated value for PM2.5 and PM10 and variations around MSB*256 which are like noise.
MC

ricki-z, 2018-06-21:
Please look at the code.
Line 67 to 72: divide values by 10 if checksum is okay and serial packet is complete.

ricki-z, 2018-06-21:
This way was chosen to save 4 bytes of memory. Saving the bytes in float variables needs 4 bytes, in integer values that needs 2 Bytes.

mcanu44, 2018-06-21:
But the result is not the same... no ?
You divide the whole value "pm10_serial" by 10, which is equal to (MSBx256+LSB)/10, thus the results is not equal to (MSB*256+LSB/10)

ricki-z, 2018-06-21:
Really??? Look at the code, then look at the sensor documentation (version 1.3)!!!
Its page 5:
PM2.5 value: PM2.5 (μg /m3) = ((PM2.5 High byte *256) + PM2.5 low byte)/10
PM10 value: PM10 (μg /m3) = ((PM10 high byte*256) + PM10 low byte)/10
Look at the brackets. Its ((MSB*256)+LSB)/10.

mcanu44, 2018-06-21:
Ah!!! Yes, you're right ;) 
Sorry.

closed 2018-06-29 by ricki-z