ricki-z/SDS011 #21: Issue after sleep function

LzkCristian, 2019-03-15:
Hello, im doing a project with DHT22 and SDS011, its a typical project as I can see.
First of all I want to thank you because of you work to make this for compatible with arduino.
Here's my issue:
I was reading the sds011 data and i havent the slept function in the code and everything was fine
then i added this line to the code
mySDS.sleep();
delay(6000);
mySDS.wakeup();
in the loop after measuring all the data, and now the fan doesnt turn on and it displays 0 as values and 1 as the "error" value, which will come as a "1" if there's no data available, and thats because its under sleepmode i think? but i cant make it to wake up...
any tips on this? 
thanks in advance!

LzkCristian, 2019-03-15:
i just figured out that if i connect it again; it works until the part of the code it says to sleep and then doesnt turn on anymore after i reconnect it again. i deleted that part and could make it work again, but my question is: how can i use the sleep mode and wakeup mode properly? i know it has been asked before but i couldnt find an answer. thanks!

ricki-z, 2019-03-15:
This seems like a problem with the SDS011. Normally neither sleep() or wakeup() are working because of a bad connection between the cpu board and the SDS011. In this case the SDS011 can send to the board, but can't receive the commands.
The wakeup() "command" for the SDS011 is very simple. As soon as a signal is send the SDS011 should start. You could even send a stop command ... So your SDS011 seems to have a problem.
But to be sure, is the fan stopping on the sleep() command and starting on the wakeup() command?

LzkCristian, 2019-03-15:
Thanks for your fast reply Ricki. The fan is stopping at the sleep command. But it's not starting at the wakeup one. What I had to do is to delete the sleep from the code and then re upload it and reconnect it (with this I mean plug off and on again) and it starts again.

ricki-z, 2019-03-15:
As mentioned the SDS011 should start again on any command sent. So this seems to be an issue with your SDS011.

LzkCristian, 2019-03-15:
Ok. I will try with another one. But I will also try wiring again, as I have shared grounds and the 5v for other sensors, maybe this could be it?
I have the tx wired to digital 6 and the RX to digital 5, and have defined them on the code. Thanks again for your fast response, Ricki. I will be posting what happens.

LzkCristian, 2019-03-15:
uhmmm nope, ive changed the wiring and also tried other SDS011 but the same happens. do you have another idea what could be happening? it might be an issue of communication, but i dont know nothing about this. thanks in advance!

hdahle, 2019-04-04:
> As mentioned the SDS011 should start again on any command sent. So this seems to be an issue with your SDS011.

I am seeing the same things. My new batch of SDS011's no longer wake-up based on the simplified sensor.wakeup() function in the library. The sensor.wakeup() function simply sends an 0x01 byte to the sensor, which *used to* wake up my SDS sensors. However that no longer works...

In the current datasheet, it is specified that the proper wake-up procedure is to send a 19-byte sequence very similar to the SLEEP command. If you modify "sds011.cpp" by adding the struct below and changing the wakeup function to be as I have specified below then everything will work.

Here's a link to the spec, refer to pages 7 and 8: [https://nettigo.pl/attachments/415] 

static const byte WAKECMD[19] = {
	0xAA,	// head
	0xB4,	// command id
	0x06,	// data byte 1
	0x01,	// data byte 2 (set mode)
	0x01,	// data byte 3 (wake up)
	0x00,	// data byte 4
	0x00,	// data byte 5
	0x00,	// data byte 6
	0x00,	// data byte 7
	0x00,	// data byte 8
	0x00,	// data byte 9
	0x00,	// data byte 10
	0x00,	// data byte 11
	0x00,	// data byte 12
	0x00,	// data byte 13
	0xFF,	// data byte 14 (device id byte 1)
	0xFF,	// data byte 15 (device id byte 2)
	0x06,	// checksum
	0xAB	// tail
};
// --------------------------------------------------------
// SDS011:wakeup
// --------------------------------------------------------
void SDS011::wakeup() {
	for (uint8_t i = 0; i < 19; i++) {
		sds_data->write(WAKECMD[i]);
	}
	sds_data->flush();
	while (sds_data->available() > 0) {
		sds_data->read();
	}
}

ricki-z, 2019-04-04:
I will change this. The "short" version was used to save some memory.

LzkCristian, 2019-04-07:
> > As mentioned the SDS011 should start again on any command sent. So this seems to be an issue with your SDS011.
> 
> I am seeing the same things. My new batch of SDS011's no longer wake-up based on the (simplified) wakeup() function in the library. The wakeup() function simply sends an 0x01 byte to the SDS011, which _used to_ wake up my SDS sensors. However that no longer works...
> 
> Solution:
> 
> In the current datasheet, it is specified that the proper wake-up procedure is to send a 19-byte sequence very similar to the SLEEP command. If you modify "sds011.cpp" by adding the struct below and changing the wakeup function similar to what I tried below then everything seems to work just fine!
> 
> Here's a link to the spec, refer to pages 7 and 8: [https://nettigo.pl/attachments/415]
> 
> ```
> static const byte WAKECMD[19] = {
> 	0xAA,	// head
> 	0xB4,	// command id
> 	0x06,	// data byte 1
> 	0x01,	// data byte 2 (set mode)
> 	0x01,	// data byte 3 (wake up)
> 	0x00,	// data byte 4
> 	0x00,	// data byte 5
> 	0x00,	// data byte 6
> 	0x00,	// data byte 7
> 	0x00,	// data byte 8
> 	0x00,	// data byte 9
> 	0x00,	// data byte 10
> 	0x00,	// data byte 11
> 	0x00,	// data byte 12
> 	0x00,	// data byte 13
> 	0xFF,	// data byte 14 (device id byte 1)
> 	0xFF,	// data byte 15 (device id byte 2)
> 	0x06,	// checksum
> 	0xAB	// tail
> };
> // --------------------------------------------------------
> // SDS011:wakeup
> // --------------------------------------------------------
> void SDS011::wakeup() {
> 	for (uint8_t i = 0; i < 19; i++) {
> 		sds_data->write(WAKECMD[i]);
> 	}
> 	sds_data->flush();
> 	while (sds_data->available() > 0) {
> 		sds_data->read();
> 	}
> }
> ```

THANKS a lot!! it worked :)

meekm, 2019-08-29:
Please update the library with the new wakeup() function as specified in the specs. I did some tests several  SDS sensors with the following serial numbers:

5002-2525  (wakeup ok)
5002-BC4E (wakeup not ok)
5002-BD44 (wakeup not ok)
5002-BDA7 (wakeup not ok)

It seems that the old SDS sensors worked fine with the short wakeup(), but the new SDS sensors needs the full qualified message of 19 bytes as is specified in the documentation.
So implement the wakeup code as is shown above.

Jupter1, 2019-09-01:
My SDS needed the changings in the .cpp file as described above as well! Now it works. Thank you!

Serial number: 5002-C8F5.