ricki-z/SDS011 #23: Time required after the wakeup call

rez39, 2019-06-29:
Hey,
Do i need to wait for few seconds after the wakeup() call, because i am getting somewhat less values when compared with no sleep command, and also according to the data sheet it shows that the response time is around 10 does that mean i need to wait for 10 seconds before reading my data from sds?

hdahle, 2019-06-29:
Yes you should wait. I wait 20 sec after waking up the device to be on the
safe side.

rez39, 2019-06-29:
i mean can i get concrete evidence that i need to use like from the data sheet or something. There was a line in the datasheet quoting it here "you can start the sensor for 30 seconds per minutes"

ricki-z, 2019-06-29:
The recommended warm up time for the SDS011 is 30 seconds. This time is needed to get a constant airflow after startup. And there seems to be done some averaging. PM isn't homogenous so you would have very different readings without averaging.
As the 30 seconds should also work for an air pipe length of 1m it should be okay to lower this time for shorter pipe. We use 15 seconds in our project luftdaten.info.

rez39, 2019-06-29:
okay thanks!

closed 2019-06-29 by rez39