ricki-z/SDS011 #24: Both examples give Error compiling for board

Paul760, 2019-09-18:
I tried to use both examples with arduino IDE with a Wemos D1 mini with a SDS011.

But where most examples work 'out of the box' both these examples do not work and do give an error during compiling. What do I have to change?

The hardware serial example script gives this error:

> /Arduino/190918_SDS011_ricki_Zschiegner_hardware_serial/190918_SDS011_ricki_Zschiegner_hardware_serial.ino: In function 'void setup()':
> 
> 190918_SDS011_ricki_Zschiegner_hardware_serial:19:16: error: 'port' was not declared in this scope
> 
>   my_sds.begin(&port);
>
>                ^
>
> exit status 1
>
> 'port' was not declared in this scope

If I remove &port I get other errors...

With the software serial script I get the following error during compiling:
>/Arduino/libraries/SDS011-master/SDS011.cpp: In member function 'void SDS011::begin(HardwareSerial*, int8_t, int8_t)':
/Arduino/libraries/SDS011-master/SDS011.cpp:118:46: error: invalid conversion from 'int8_t {aka signed char}' to 'SerialMode' [-fpermissive]
  serial->begin(9600, SERIAL_8N1, rxPin, txPin);
>                                              ^
>In file included from >/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Arduino.h:263:0,
                 from /Arduino/libraries/SDS011-master/SDS011.h:12,
                 from /Arduino/libraries/SDS011-master/SDS011.cpp:11:
/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h:87:10: error:   initializing argument 3 of 'void HardwareSerial::begin(long unsigned int, SerialConfig, SerialMode, uint8_t)' [-fpermissive]
     void begin(unsigned long baud, SerialConfig config, SerialMode mode, uint8_t tx_pin);
>          ^
>exit status 1
>Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

bill-orange, 2019-09-21:
@Paul760 

Try this:  Modify SDS011.cpp by adding an iefdef and endif, like this:

```
#ifdef ESP32
void SDS011::begin(HardwareSerial* serial, int8_t rxPin, int8_t txPin) {
	serial->begin(9600, SERIAL_8N1, rxPin, txPin);
	sds_data = serial;
}
#endif
```

Let me know if that fixes it.

Bill

tikh-iv, 2019-12-17:
@bill-orange 
Thank you! It's work. But I don't understand why.

bill-orange, 2019-12-17:
glad it worked!