ricki-z/SDS011 #25: Can not compile SDS011_example for ESP8266

bill-orange, 2019-09-21:
The ESP32 example compiled fine for ESP32 boards, at lease in my environment of the Ardoino libraries and IDE.

The SDS011_example would not compile for any ESP8266 board I tested.

> Arduino: 1.8.9 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"
> 
> Build options changed, rebuilding all
> C:\Users\William\Documents\Arduino\libraries\SDS011-master\SDS011.cpp: In member function 'void SDS011::begin(HardwareSerial*, int8_t, int8_t)':
> 
> C:\Users\William\Documents\Arduino\libraries\SDS011-master\SDS011.cpp:118:46: error: invalid conversion from 'int8_t {aka signed char}' to 'SerialMode' [-fpermissive]
> 
>   serial->begin(9600, SERIAL_8N1, rxPin, txPin);
>                                               ^
> In file included from C:\Users\William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Arduino.h:263:0,
> 
>                  from C:\Users\William\Documents\Arduino\libraries\SDS011-master\SDS011.h:12,
> 
>                  from C:\Users\William\Documents\Arduino\libraries\SDS011-master\SDS011.cpp:11:
> 
> C:\Users\William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h:87:10: error:   initializing argument 3 of 'void HardwareSerial::begin(long unsigned int, SerialConfig, SerialMode, uint8_t)' [-fpermissive]
> 
>      void begin(unsigned long baud, SerialConfig config, SerialMode mode, uint8_t tx_pin);
> 
>           ^
> 
> Multiple libraries were found for "SoftwareSerial.h"
>  Used: C:\Users\William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial
>  Not used: C:\Users\William\Documents\Arduino\libraries\EspSoftwareSerial
> exit status 1
> Error compiling for board NodeMCU 0.9 (ESP-12 Module).
> 

Thoughts?

zoomx, 2020-09-03:
The same for me

sketch\SDS011.cpp: In member function 'void SDS011::begin(HardwareSerial*, int8_t, int8_t)':
SDS011.cpp:118:46: error: invalid conversion from 'int8_t {aka signed char}' to 'SerialMode' [-fpermissive]
  serial->begin(9600, SERIAL_8N1, rxPin, txPin);
                                              ^
In file included from D:\Programmi\arduino\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/Arduino.h:245:0,
                 from sketch\SDS011.h:12,
                 from sketch\SDS011.cpp:11:
D:\Programmi\arduino\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h:87:10: error:   initializing argument 3 of 'void HardwareSerial::begin(long unsigned int, SerialConfig, SerialMode, uint8_t)' [-fpermissive]
     void begin(unsigned long baud, SerialConfig config, SerialMode mode, uint8_t tx_pin)
          ^
exit status 1
invalid conversion from 'int8_t {aka signed char}' to 'SerialMode' [-fpermissive]

bill-orange, 2020-09-03:
@zoomx   I just tested with the following patch and it compiled for me.  The patch to SDS011.cpp looks like this:

```
#ifndef ESP32
void SDS011::begin(uint8_t pin_rx, uint8_t pin_tx) {
	_pin_rx = pin_rx;
	_pin_tx = pin_tx;

//WEW
//	SoftwareSerial *softSerial = new SoftwareSerial(_pin_rx, _pin_tx);
    SoftwareSerial *softSerial = new SoftwareSerial();
//	softSerial->begin(9600);
    softSerial->begin(9600,SWSERIAL_8N1,_pin_rx, _pin_tx,false,32);

	sds_data = softSerial;
}
#endif

```

zoomx, 2020-09-03:
Thanks! I have made this modifications and it compiles but my sensor is tied to airrohr firmware but I will test it.