ricki-z/SDS011 #33: version 0.0.6 ?

AugustQu, 2020-02-17:
I assume that the confusion with my code here is based on the version-info in the file libray.properties. It says: 0.0.6
PlatformIO fetches a library with the same version-number but it's an older one but I need the newer one.

Can you please change the version-info in library.properties to 0.0.7.

Thanks.

cyberman54, 2020-02-25:
Problem here is that for the platformio crawler a github release or tag is necessary.
Currently release is 0.0.5, but library.properties says 0.0.6.

-> Please make git release for v0.0.6, this will solve the issue.

JackGruber, 2020-02-26:
you can use the git url instead of the name in PlatformIO to get the latest version from git.

```
lib_deps =
          https://github.com/ricki-z/SDS011
```

cyberman54, 2020-02-26:
@JackGruber Yes, this works, but is not the intention of how to use platformio registry.