ricki-z/SDS011 #35: no sleep()/wakeup() for ESP32

AugustQu, 2020-03-26:
Hi,

I'm working with a board containing the ESP32 plus this library. I can receive values for PM2.5 and PM10, but I cannot put the SDS011 to sleep.

Here is a simplified version of my code:
```
void setup()
{
    Serial.begin(115200);
    sdsSensor.begin(&sdsSerial, SDS_PIN_RX, SDS_PIN_TX);
    Serial.println("SDS011 start/stop and reporting sample");
}

void loop()
{
    Serial.println("stop SDS");
    sdsSensor.sleep();
    delay(5000);
    Serial.println("start SDS");
    sdsSensor.wakeup();
    delay(5000);    
    int sdsErrorCode = sdsSensor.read(&pm25, &pm10);
}

```
Any tips?
```

AugustQu, 2020-03-26:
adding some print-statements to this code it outputs this:
```
start SDS
0 9.10 4.10
stop SDS

```
the numbers have this meaning:
the returncode of the read() is 0
PM10 has a value of 9.10
PM2.5 has a value of 4.10

So it looks like the sensor is working.

But the sleep() does nothing.

JackGruber, 2020-03-26:
Hi,

below is a copy paste of a code I use.
After the wakeup you have to wait 30 seconds.

```
  sds.begin();
  sds.setQueryReportingMode(); // ensures sensor is in 'query' reporting mode  
  

  sds.wakeup();
  delay(30000); for warm up sensor
  PmResult pm = sds.queryPm();
  if (pm.isOk()) 
  {
    Serial.print("PM2.5 = ");
    Serial.print(pm.pm25);
    pm25 = pm.pm25;
    Serial.print(", PM10 = ");
    pm10 = pm.pm10;
    Serial.println(pm.pm10);
  } 
  else 
  {
    Serial.print("Could not read values from sensor, reason: ");
    Serial.println(pm.statusToString());
  }

  WorkingStateResult state = sds.sleep();
  if (state.isWorking()) 
  {
    Serial.println("Problem with sleeping the sensor.");
  } 
  else 
  {
    Serial.println("Sensor is sleeping");
  }
 
```

ricki-z, 2020-03-26:
Please check the pins the SDS011 ist connected to and that these are the same than in the definition (SDS_PIN_RX, SDS_PIN_TX). And check that the connecting cable between the ESP32 and the RX pin of the SDS011 is okay.
It seems like the SDS011 can send data (via TX pin) but isn't receiving the commands (via RX pin).

AugustQu, 2020-04-21:
I have an idea:

in the code the _wakeup()_-function is realized as:
`void SDS011::wakeup() {
  sds_data->write(0x01);
  sds_data->flush();
}
`
For me this worked on an Arduino Uno and also on an Arduino Mega 2560, but it did not work on an ESP32-board. 

For my tests I used the little program I presented in the beginning of this thread. On the ESP32-board the SDS011 starts working, put to sleep but would never wakeup again.

I modified the source code into this form:
`void SDS011::wakeup() {
//  sds_data->write(0x01);
//  sds_data->flush();
  for (uint8_t i = 0; i < 19; i++) {
    sds_data->write(SDS_START[i]);
  }
  sds_data->flush();
  while (sds_data->available() > 0) {
    sds_data->read();
  }
}
`
and in the beginning of the file I included these lines:
`static const byte SDS_START[] = {
  0xAA, 0xB4, 0x06, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0xFF, 0x06, 0xAB};
`

Et voila: now it works.

Can someone please verify my result?

PS: this is the board I'm using:
https://de.aliexpress.com/item/32915894264.html

x-RoNiN-x, 2020-04-21:
@AugustQu 
Hello August.
This code works fine for a week now ...
Our pax counter works well with TTN BME280 / 680 SDS011.

We use the liligo in a leisure facility and at my home as a kind of weather station! Can we somehow get in touch?

I think we have the same goal ...

the times of the paxcounter are not right yet!

best regards

Daniel

AugustQu, 2020-04-22:
@x-RoNiN-x 
do you work with the original code? Or did you include the modifications regarding the function wakeup()?

Here on my side my test-code now works but only after I apply the modifications. And it also works on the paxcounter-side (= with the full software of the paxcounter).

Maybe it's a problem with the board I'm using. Nevertheless I think it's a good idea to use the "official" command-string instead of this mighty trick.

x-RoNiN-x, 2020-04-22:
@AugustQu 
There are two types of SDS. can be recognized by the serial number. once only with numbers, and the others with numbers and letters. The SDS with letters in the serial number does not work with the standard software. I've been running a pax counter with an older version of the SDS for several weeks. However, the SDS starts up every 30 seconds. this is of course not energy efficient. The question is whether something can be changed in the code so that the particulate matter sensor runs more efficiently ....

Best regards

AugustQu, 2020-04-23:
@x-RoNiN-x 
Thanks for your hint. Looking at the case of the sensor I see something that looks like a serial number. It contains 2 blocks, and the last block contains the letter A. So it looks like a got one of this series.

Nevertheless I would prefer having a software that works in both cases. So for me it would be the best approach to use the 'official wakeup-string'.

cyberman54, 2022-02-14:
@AugustQu can we have a PR for paxcounter for this?

GFMsoft, 2023-02-19:
I am working with a [ESP32 NodeMCU](https://www.az-delivery.de/products/esp32-developmentboard).
Had the same problems with the wakeup.

Changed the .cpp like @AugustQu commented on Apr 21, 2020 and it worked. 
Thank you.